adafruit/Adafruit_Python_PCA9685 #1: I can't get the drivers to install on my Raspberry Pi. If I use pip install I get:

helloman1802, 2016-09-01:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

helloman1802, 2016-09-01:
When I use pip install to install the drivers I get:
`
Downloading/unpacking adafruit-pca9685
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1178, in prepare_files
    url = finder.find_requirement(req_to_install, upgrade=self.upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 194, in find_requirement
    page = self._get_page(main_index_url, req)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 568, in _get_page
    session=self.session,
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 694, in get_page
    req, link, "connection error: %s" % exc, url,
TypeError: __str__ returned non-string (type Error)

Storing debug log for failure in /root/.pip/pip.log

If I use setup.py it tells me that it is missing a file and that I might be missing some files.
`

Thanks,
Malachi

tdicola, 2016-10-19:
It looks like the version of pip you have installed might have issues.  I would start from a fresh install of Raspbian, then get the latest pip by running:
```
sudo apt-get update
sudo apt-get install python-dev python-pip
sudo pip install --upgrade pip
```
Give that a shot to see if it helps fix the pip issue.  I can't repro the problem when I try on a current Raspbian build:
```
pi@raspberrypi:~/Adafruit_Python_MCP4725 $ sudo pip install adafruit-pca9685
Downloading/unpacking adafruit-pca9685
  Downloading Adafruit_PCA9685-1.0.0-py2-none-any.whl
Requirement already satisfied (use --upgrade to upgrade): Adafruit-GPIO>=0.6.5 in /usr/local/lib/python2.7/dist-packages/Adafruit_GPIO-1.0.0-py2.7.egg (from adafruit-pca9685)
Requirement already satisfied (use --upgrade to upgrade): adafruit-pureio in /usr/local/lib/python2.7/dist-packages/Adafruit_PureIO-0.2.0-py2.7.egg (from Adafruit-GPIO>=0.6.5->adafruit-pca9685)
Requirement already satisfied (use --upgrade to upgrade): spidev in /usr/local/lib/python2.7/dist-packages/spidev-3.2-py2.7-linux-armv7l.egg (from Adafruit-GPIO>=0.6.5->adafruit-pca9685)
Installing collected packages: adafruit-pca9685
Successfully installed adafruit-pca9685
Cleaning up...
```

closed 2016-10-19 by tdicola