adafruit/Adafruit_Python_PCA9685 #9: importerror no module named 'adafruit pca9685

orkutsoft, 2018-08-03:
i am using putty and remote desktop to connect to raspberry pi
when i run program with Adafruit_PCA9685 in putty command line that uses python 2.7 it runs the program but when when i use python 3.5.3 shell it gives import error
i did every thing this conversation says
but
SAME ERROR
"importerror no module named 'adafruit pca9685'"

Georgielaa, 2018-08-24:
When you installed the adafruit library and ran the setup with :- sudo python setup.py install, this only sets it up for python 2. If you want to use the library with python3 you have to run this setup.:- sudo python3 setup.py install

ladyada, 2018-08-24:
hiya, wanna try our circuitpython version of this library? it works for sure on raspi and we have detailed instruction here:
https://learn.adafruit.com/16-channel-pwm-servo-driver/python-circuitpython
this library will be deprecated

closed 2019-10-09 by ladyada