adafruit/Adafruit_Python_PCA9685 #16: Issues communicating via I2C: BBB --> PCA9685

bRitch022, 2019-07-10:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Linux beaglebone 4.14.71-ti-r80

- Python version (run `python -version` or `python3 -version`):  Python 2.7.13

- Error message you are receiving, including any Python exception traces:  
Traceback (most recent call last):
  File "/var/lib/cloud9/Adafruit_Python_PCA9685/examples/simpletest.py", line 17, in <module>
    pwm = Adafruit_PCA9685.PCA9685()
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_PCA9685-1.0.1-py2.7.egg/Adafruit_PCA9685/PCA9685.py", line 75, in __init__
    self.set_all_pwm(0, 0)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_PCA9685-1.0.1-py2.7.egg/Adafruit_PCA9685/PCA9685.py", line 111, in set_all_pwm
    self._device.write8(ALL_LED_ON_L, on & 0xFF)
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/I2C.py", line 116, in write8
  File "build/bdist.linux-armv7l/egg/Adafruit_PureIO/smbus.py", line 256, in write_byte_data
IOError: [Errno 16] Device or resource busy

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):
Using BBB channel I2C-2 (P9_19 & P9_20), attach I2C connections to PCA9685 development board.
Using Adafruit_Python_PCA9685 library, run example program: simpletest.py

closed 2019-07-31 by bRitch022

asteinig4018, 2019-08-10:
I've run into a related problem with the same error. Would you mind explaining how your resolved this?

bRitch022, 2019-08-12:
I resolved this by selecting the correct bus in the simpletest.py file. On my BBB, it was bus 2.
Uncomment the following line, and make sure that the address is 0x40, unless you have manually soldered the address offset on the PCA9685 dev board.
```
# Alternatively specify a different address and/or bus:
#pwm = Adafruit_PCA9685.PCA9685(address=0x41, busnum=2)
```

bRitch022, 2019-08-12:
> I've run into a related problem with the same error. Would you mind explaining how your resolved this?

I resolved this by selecting the correct bus in the simpletest.py file. On my BBB, it was I2C bus 2 because I am using pins 9_19 and 9_20.
Uncomment the following line, and make sure that the address is 0x40, unless you have manually soldered the address offset on the PCA9685 dev board.

# Alternatively specify a different address and/or bus:
#pwm = Adafruit_PCA9685.PCA9685(address=0x41, busnum=2)